MelbourneHighSchoolRobotics/ev3sim #55: Dragging pygame window kills ongoing `attach` scripts, if waiting on data.

glipR, 2020-08-28:
Because dragging the pygame window pauses execution on that thread, no tick update data is sent which causes `Simulation Died` in the communications threads, if waiting for a send/recv.

Unsure what the best way to fix this is.

peter-drew, 2020-08-29:
@Dalordish @angustrau thoughts?

Dalordish, 2020-08-29:
Going to pull down and have a try at some possible fixes I've seen on stackoverflow in a bit. The best  solution would be to figure out if we can get pygame to stop donig that, but barring that, possible workarounds include:
- It seems relatively trivial to detect when a window move event happens, so we could just pause the game when that happens - not sure if that would require modifications to the way the robot clients interact with the server though, or if the server would be able to send out "game paused" before the thread froze
- This is undesireable, but would probably work - the simulation happens on a different thread to the graphics display/pygame window, so the simulation continues running just fine

glipR, 2020-08-29:
https://github.com/pygame/pygame/issues/1525 seems like this is a limitation of SDL, which pygame uses.

Doesn't seem feasible to change the way pygame operators, so I think what needs to happen is the communication thread dies out when it DEFINITELY knows the simulation thread has exited (just send thru some information on a shared queue), as opposed to after no data has been received for a certain amount of time.

PR #61 (2020-08-29): Bug window move
Fixes #55 .

Was a lat more painful than I was expecting, turns out that the attach script was also silently raises a Deadline exception, which needed to be handled separately.

Also added support for dynamic resizing of the window, which while technically functional doesn't handle a few of the objects in our soccer preset to nicely (Try it out for yourself to get an idea of this)
closed 2020-08-29 by glipR