MelbourneHighSchoolRobotics/ev3sim #59: Enforce better codestyle

glipR, 2020-08-29:
The more I look at this codebase the more I'm noticing inconsistencies in codestyle which probably should be unified, and checked for new PRs. I remember @Dalordish was using `black` on the MindPile project, so maybe we add that to this project? Otherwise open to other options.

Dalordish, 2020-08-29:
Would reccomend `black` as a simple, opinionated way to get automatic code formatting and style enforcement. If you/people have different sensibilites though, `yapf` is more configurable, and` autopep8` doesn't do full auto code formatting, but at least enforces PEP8 as a minimum standard.

Dalordish, 2020-08-29:
@angustrau @peter-drew any opinions?

peter-drew, 2020-08-29:
I'd probably prefer something that fails a check and forces the user to make changes, rather than do silently in the background, as more educational for students. But that would get annoying if it was too pedantic.

Maybe it autocorrects small stuff, but fails on larger stuff. When it makes changes, is it guaranteed to not change program flow or otherwise corrupt stuff @Dalordish?

Dalordish, 2020-08-29:
IMO this fits neatly into the "do we have minimum code standards ala PEP8, or do we have a single, enforced, automatic, set and forget code styler" question. 

We can enforce PEP8 pretty easily with flake8 in our CI/CD suite, which means you won't be able to merge unless you're conforming.

Another option is the more complicated route of setting things up with commit hooks, so all code gets linted and formatted on commit.

Dalordish, 2020-08-29:
I'm not sure how much educational value there is in forcing students to make the changes manually - most of the code style stuff we're talking about is pedantic things like operator spacing (a + b, not a+ b), or line length. There's probably a marginal amount there? Making sure people get in the habit even when they don't have an automated tool? But on the flipside, setting up and getting used to an automated tool has some educational value in exposure.

glipR, 2020-08-29:
I personally think workflows should lint, not change the code. But I agree with Richard that not automating the formatting itself doesn't serve much purpose. I think students get the same amount of experience by writing code as they wish and then running `black` and the end of it, in a separate commit - That way they still easily see what changes are made in GH desktop or smth.

peter-drew, 2020-08-29:
I tend to agree. Personally I wouldn't want it to handle newlines, but that is personal preference more than anything else (also I think there are some issues in Django with automatic newlines for example, but not an issue here). Could probably set this up to run on the master branch such that it only happens after the pull request is merged.

glipR, 2020-08-29:
Maybe I'm misunderstanding, but my interpretation was that we lint with something like black on pull request targets - IE you can't merge until the linting passes, but you only need to run `python -m black .` locally then push the changes to fix that.

Do you think it would be better to just have the merged branches be automatically formatted? I guess the end result is the same but in the latter we avoid a bit of the tedium. I'm good with either approach.

glipR, 2020-09-12:
Closing since for the most part resolved by #102

closed 2020-09-12 by glipR