MelbourneHighSchoolRobotics/ev3sim #60: Relative imports failing with ev3attach

glipR, 2020-08-29:
Another thing that's been brought to my attention is that relative imports don't work with ev3attach.

I've been messing with importlib for a while now but haven't got too far. Here's my current progress on changing some of the code in `attach.py`, although this doesn't fix the issue:

```
                import os
                from importlib.machinery import SourceFileLoader
                from importlib.util import spec_from_loader, module_from_spec
                class ModLoader(SourceFileLoader):
                    def get_filename(self, fullname):
                        if self.path.endswith('.py'):
                            return self.path
                        return os.path.join(self.path, '__init__.py')
                loader = ModLoader('__main__', fname)
                spec = spec_from_loader(loader.name, loader)
                module = module_from_spec(spec)
                loader.exec_module(module)
```

platy11, 2020-08-29:
Can you just mock `sys.path`? This seems to work for me, but it probably causes issues that I'm not considering
```diff
diff --git a/ev3sim/attach.py b/ev3sim/attach.py
index d78b029..d928653 100644
--- a/ev3sim/attach.py
+++ b/ev3sim/attach.py
@@ -8,6 +8,7 @@ import time
 import argparse
 from unittest import mock
 from queue import Queue
+from os import path
 
 def main(passed_args = None):
     if passed_args is None:
@@ -303,6 +304,9 @@ def main(passed_args = None):
                     info = data['write_results'].get()
                     data['active_connections'].remove(self)
 
+            fakepath = sys.path.copy()
+            fakepath.append(path.dirname(filename))
             @mock.patch('time.time', get_time)
             @mock.patch('time.sleep', sleep)
             @mock.patch('ev3dev2.motor.Motor.wait', wait)
@@ -312,6 +316,7 @@ def main(passed_args = None):
             @mock.patch('ev3sim.code_helpers.is_sim', True)
             @mock.patch('ev3sim.code_helpers.CommServer', MockedCommServer)
             @mock.patch('ev3sim.code_helpers.CommClient', MockedCommClient)
+            @mock.patch('sys.path', fakepath)
             def run_script(fname):
                 from importlib.machinery import SourceFileLoader
                 module = SourceFileLoader('__main__', fname).load_module()
```

glipR, 2020-08-29:
Good idea. Will have a bit of a think but provided nothing more stable comes along I think this is a good change.

glipR, 2020-08-29:
I've made a slight change in the PR: favouring the cwd we invoked the script from, rather than the parent of the code being run. This allows for a bit smart imports (we can import code from separate folders, as long as they are in the `cwd` and reference relative to the `cwd`.

PR #76 (2020-08-30): Add invoking folder to mocked path for attach scripts.
Fixes #60 .
closed 2020-08-30 by glipR