MelbourneHighSchoolRobotics/ev3sim #78: Sensor bias and randomisation

peter-drew, 2020-09-02:
If we want to implement this for the senior division we need to discuss how this will be done and start work soon.

peter-drew, 2020-09-03:
This looks good. A few small suggestions:

MOTORS:
- We probably want to define the algorithm in a way that the bias is consistent between runs - because in real life the bias is tied to the physical motor. This also applies to other sensors with bias for that matter.

ULTRASONIC:
- Angle sounds good. Probably also want to just add a little random noise, at all angles.

COMPASS:
- The most annoying thing about real compasses is the non-linear rate change as the compass is rotate - ie rotating the bot 90 degrees might result in a 70 degree change, then the next 90 degrees results in 110. Implementing this would make the compasses very realistic, possibly to the point where it becomes too difficult. So would want the amount of bias and inaccuracy to be adjustable (probably the case for all sensors).

glipR, 2020-09-03:
Good points.

I think regarding biases that stay the same between runs, we can have the option to randomise a value based off of one of two things:
* The world seed (IE different every time)
* The filename of the bot + the port of the sensor (And if there are multiple of the same filename, the index of this filename)

This would mean that if the port and filename stay the same, the values should too.

Agreed on the ultrasonic. Regarding the compass (and all other sensor values) there are pretty easy ways to implement some modularity with their application:

* `n` stable values can change the values `n`, as well as the variability of distribution of these stable values
* Initial bias can be multiplied by some variability constant
* random noise can also have some multiplication occurring

Regarding random noise, do we want something smooth like perlin noise, or would we prefer using `random.random()`? While you'd expect some continuity between error, it also seems quite erratic on the devices.

peter-drew, 2020-09-03:
I think the world seed and filename + port are both good sources and we might potentially use both for different things. For example, bias of the compass would depend on the environment, so would use world seed. Bias of motors would use the filename + port.

Regarding random noise, I think random.random() is probably fine. It may not be an entirely accurate representation, but it is probably easier for students to remove by taking a rolling average, but still provide enough of a challenge.

peter-drew, 2020-09-24:
Closed by #82

closed 2020-09-24 by peter-drew