MelbourneHighSchoolRobotics/ev3sim #85: Evaluate rotational speed and acceleration constants

peter-drew, 2020-09-10:
Possibly too fast at the moment

peter-drew, 2020-09-24:
I think especially of the ball

peter-drew, 2020-09-24:
Happy to take suggestions on values and either resolve or close relatively quickly.

glipR, 2020-09-24:
re: ball, are you concerned with rotational acceleration or just general friction? I haven't noticed the ball rotating, since its a solid red :/

The angular and normal velocity are dampened by me in ev3sim/objects/base.py line 121->125, so you can play around with the constants mentioned there + add your own to fine tune the rotational speeds.

peter-drew, 2020-09-25:
> re: ball, are you concerned with rotational acceleration or just general friction?

General friction. I'll take a look.

PR #149 (2020-09-25): Adjust constants
Closes #85
closed 2020-09-25 by peter-drew