MelbourneHighSchoolRobotics/ev3sim #100: Add ability for variable match length and 2 halves

peter-drew, 2020-09-11:
Currently resetting the match clock also resets the score. Would be good to have some way of having a second half and swapping the direction the teams are shooting.

Would also be nice to make match length variable (in preset is fine).

If scores a level time on is sometimes also needed.

glipR, 2020-09-11:
Provided https://github.com/MelbourneHighSchoolRobotics/ev3sim/pull/94 is merged you've already got variable time. I think once 5mins is done there's a quick 5 second pause then it automatically swaps sides as part of the judge.

That being said they should probably still be a ui button to achieve the same effect.

peter-drew, 2020-09-24:
I don't think the side swapping exists or works as is. And is probably needed before we get to running the competition.

Unless we choose to not swap sides. Would need a way for robots to recalibrate their compass. Current advice to teams is just do it on startup and I am reluctant to do anything that complicates that, especially at this stage, especially given I don't think there currently exists a hook that teams could use.

Given there is no external physical interference, no side on the field has an advantage so the need to swap sides is reduced. And the need to repair physical robots in half time is also reduced.

Thoughts @Dalordish @glipR ?

peter-drew, 2020-09-24:
That being said, might want to reconsider for 2021.

glipR, 2020-09-24:
I tend to agree re: fairness, but I think we can make a relatively simple change to get this to work - Make the simulator able to reset a bots code. I'm pretty sure this wouldn't take too much effort, then we could bind it to a button in the ui, so you'd pause the game, swap halves, reset the bots, then begin play.

glipR, 2020-09-24:
That being said I agree its probably not priority.

peter-drew, 2020-09-25:
> Make the simulator able to reset a bots code.

I think this is the only way we could implement this without causing problems for students.