MelbourneHighSchoolRobotics/ev3sim #116: Infrared sensor inconsistency with ev3dev

platy11, 2020-09-16:
The simulated InfraredSensor, when in 'all values (modulated)' mode (`AC-ALL`) returns [7 values](https://github.com/MelbourneHighSchoolRobotics/ev3sim/blob/87dc4becd832cd057d051fe12cc4cadcecc044fa/ev3sim/devices/infrared/base.py#L73), unlike the ev3dev drivers where, according to [the documentation](https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-ir-seek-v2-modes), 6 values are returned (7 values are returned in the unmodulated mode `DC-ALL`).

I'm not sure what the best solution to this would be, since the 7th value could be used in existing programs (such as templatebot.py in the template repo, line 69).

platy11, 2020-09-16:
I encountered this issue when writing a custom `InfraredSensor` class, which includes functions that look somewhat like this:
```py
@property
def all_values(self) -> Tuple[Direction, Tuple[int, int, int, int, int]]:
    """
    InfraredSensor.all_values returns as a tuple:
        - the direction (as returned by InfraredSensor.direction_modulated) (InfraredSensor.Direction)
        - a tuple of the signal strength from each of the 5 sensors (tuple[5]: int: 0 to 9)
    The sensor is used in modulated mode.
    """
    self.mode = InfraredSensor.MODE_ALL_MODULATED
    return (
        InfraredSensor.Direction(self.value(0)),
        (
            self.value(1),
            self.value(2),
            self.value(3),
            self.value(4),
            self.value(5)
        )
    )
```
Reading `InfraredSensor.all_values_unmodulated` (which additionally returns `self.value(6)`) would change the sensor mode to `DC-ALL`, causing an exception in ev3sim. Returning a seventh value in `InfraredSensor.all_values`, or not changing the mode when a property is read would cause problems if the program was running on the robot.

glipR, 2020-09-16:
Calling `value(6)` doesn't cause errors for me as it doesn't automatically change the sensor mode to DC.

That being said in order to be an accurate simulator we should accurately simulate the values provided by the spec, so we should not return a 6th value in AC mode (idk why though, is just an easy calculation).

That being said I'm not sure what values we want to give the DC mode - this technically shouldn't see the ball right @Dalordish @peter-drew ? So should we just return 0,0,0,0,0, just to make sure students don't accidentally use DC and then get confused why it works in sim but not real life.

glipR, 2020-09-16:
Also, rather than explcitly setting the mode you should use `self._ensure_mode(modename)` this is what the internals of ev3 devices does. (So provided you are extending the sensor class you will have access to this).

peter-drew, 2020-09-16:
> So should we just return 0,0,0,0,0, just to make sure students don't accidentally use DC and then get confused why it works in sim but not real life.

Probably...? Without having an ev3 to test with I'm not sure what that mode returns with an actual ball. I don't think this has ever been an issue in the past and I was of the impression that we were able to get an average strength with our ball, but maybe that is just in Mindstorms.

platy11, 2020-09-16:
> so we should not return a 6th value in AC mode (idk why though, is just an easy calculation).

It does seem very weird that the sensor only has the average in AC mode, it was making me wonder if I was missing something and there was some reason ev3dev has it like this

> Also, rather than explcitly setting the mode you should use self._ensure_mode(modename) this is what the internals of ev3 devices does.

Yeah, I realised this earlier after I checked some other things in ev3dev-lang-python :slightly_smiling_face:

PR #119 (2020-09-17): Update IR to support all valid modes.
Fixes #116 .

Has also illuminated another issue though - we don't wait for new values after mode changes, which means values can be out of date.
closed 2020-09-17 by glipR