MelbourneHighSchoolRobotics/ev3sim #140: Move demo scripts to scripts folder

peter-drew, 2020-09-23:
https://github.com/MelbourneHighSchoolRobotics/ev3sim/tree/main/ev3sim/robots/scripts

Not a hard change, but need to update all locations that link here - including potentially outside this repository.

peter-drew, 2020-09-23:
@glipR would this affect where ev3sim looks for default scripts?

peter-drew, 2020-09-23:
Actually might be better off creating a new 'examples' folder and moving the robots, scripts and batched_command folders into it. But will require more link updates and probably code changes for the default search locations.

glipR, 2020-09-23:
At the moment that would break stuff but its an easy fix.

glipR, 2020-09-23:
I'd probably rather having a few bots in separate folders with bot yamls and python code, with batched commands one level up.

glipR, 2020-12-06:
With the new gui and how that all works, We shouldn't need to delve into the package files really at all, especially from a student's perspective. I still agree this stuff should all sit in an examples folder though.

glipR, 2020-12-06:
Fixed by #184

closed 2020-12-06 by glipR