MelbourneHighSchoolRobotics/ev3sim #189: TypeError: initializer for ctype 'struct cpVect' must be a list or tuple or dict or struct-cdata, not numpy.ndarray

logwet, 2020-12-12:
I just updated from ev3sim 1.3.3 to 1.6.5 and it isn't working anymore. The same problem occurs if I try to run it using the batched commands system.

```pytb
(2020-Simulator-Team-P) C:\Users\sachi\PycharmProjects\2020-Simulator-Team-P>ev3sim bot.yaml bot.yaml
Simulating with seed 3103703429

=====================================================
DEPRECATED: Running ev3sim without the batch argument will be removed in a newer version of ev3sim.

To find out how to use batch commands, consult https://ev3sim.mhscsr.club/batched_commands.html
=====================================================

pygame 2.0.0 (SDL 2.0.12, python 3.8.5)
Hello from the pygame community. https://www.pygame.org/contribute.html
An error occurred in the Simulation thread. Raising an error now...
Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\ProgramData\Anaconda3\envs\2020-Simulator-Team-P\Scripts\ev3sim.exe\__main__.py", line 7, in <module>
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\sim.py", line 85, in main
    single_run(args.preset, args.robots, args.bind_addr, args.seed)
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\single_run.py", line 95, in single_run
    raise r[1]
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\single_run.py", line 65, in run
    runFromConfig(config, shared_data)
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\simulation\loader.py", line 169, in runFromConfig
    sl.loadElements(config.get("elements", []))
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\simulation\loader.py", line 73, in loadElements
    obj = objectFactory(**item)
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\objects\base.py", line 140, in objectFactory
    r.initFromKwargs(**options)
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\ev3sim\objects\base.py", line 106, in initFromKwargs
    self.body.position = self.position + self.visual.getPositionAnchorOffset()
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\pymunk\_typing_attr.py", line 11, in __setattr__
    super().__setattr__(name, value)
  File "c:\programdata\anaconda3\envs\2020-simulator-team-p\lib\site-packages\pymunk\body.py", line 296, in _set_position
    lib.cpBodySetPosition(self._body, pos)
TypeError: initializer for ctype 'struct cpVect' must be a list or tuple or dict or struct-cdata, not numpy.ndarray
```

The same error occurs if I try to run with a batched flag.

logwet, 2020-12-12:
Downgraded to 1.3.3, downgraded numpy to 1.19.0 and pygame to 1.9.6, used the demo controllable.yaml file packaged with ev3sim and still having the same problem. I suspect it's due to a conflict with another package or perhaps a problem with my environment.

logwet, 2020-12-12:
Problem still occurs in a fresh venv instead of a conda env. I believe I've tried every sensible version combination of numpy, pygame and ev3sim now.

glipR, 2020-12-12:
Thanks for testing all of those configurations, and apologies for taking a bit to answer.
This seems to be a problem with `pymunk=6.0.0`. I'll probably try fix this as part of #184 , but for now you can fix it by running `pip install pymunk=5.7.0`.

logwet, 2020-12-12:
Thank you so much, this was driving me absolutely crazy.

closed 2020-12-12 by logwet