MelbourneHighSchoolRobotics/ev3sim #212: Reorder workspace format

glipR, 2021-02-15:
Allow robot code to be sectioned off within folders for individual bots (And enforce this when linking code to bots).

peter-drew, 2021-02-15:
Could we remove the need to link code at all and rather make it that the code must be in the same folder as the bot and called something like bot.py. We could have the name of the bot defined exclusively by the folder name and then inside the definition and code, and any other code that is called would also be placed there.

glipR, 2021-02-15:
Good idea

glipR, 2021-02-15:
Actually, think what might be best is a list view for code selection, similar to bot selection for sims. So when you edit a bot, you get a list view of all `.py` files in the bot folder, which you select as the script for this bot. You can also add a `+` button to create the file for you, then people won't have problems with the wrong file extensions, and if `code` is recognised in the command line we can open the workspace automatically.

glipR, 2021-02-15:
That last point won't work for portable installs, obviously.

angustrau, 2021-02-16:
I'm for having a more strict folder structure with a preset entry point. I think it would reduce confusion to have 1 documented option rather than adding more UI overhead

glipR, 2021-02-16:
Fair enough, what about best of both worlds - Bots always enter in at `bot.py`, and when a bot is created, the folder, bot file, image and `bot.py` are all created automatically. We can also add a button on the bot page to open the folder.

glipR, 2021-02-16:
One thing I'll change at the same time - Renaming bots won't break presets that reference them.

angustrau, 2021-02-16:
That sounds like a good solution

PR #222 (2021-02-22): Bot workspace change
Fixes #212 

Makes the simulator use the new workspace standard.

TODO:
Make it so that the sim automatically detects old bot versions, and alerts the user, converts the bot into new format (This will break bots with existing relative imports but meh)
closed 2021-02-22 by glipR