MelbourneHighSchoolRobotics/ev3sim #233: Add animation functions

jtbui20, 2021-02-22:
A couple that would probably be implemented just to ease the eyes:
- Position animation: (x,y) -> (x,y)
- Opacity animation: (invisible to visible)
- Colour gradient animation: (hex) -> (hex)
Probably have an animation time duration argument as well.
Some references
https://gist.github.com/zeffii/c1e14dd6620ad855d81ec2e89a859719
https://easings.net/#

glipR, 2021-02-22:
Yep love me some easing functions and animations, had way too much fun playing around with manim. Was originally thinking about implementing while I was writing v2 but got left to the wayside when I realised how much other stuff needed to be done :)

What do you think is the best way to approach this from an API perspective? Each menu has an update and draw_ui function. We could register animations of certain objects in the respective menus, and then the base `update` function modifies the object properties with respect to update's `dt` value.

Opacity and colour gradient will need to be played around with a bit. As you might've seen in some of the code, changing the theming properties of pygame_gui objects can be quite involved depending on the property (and pygame by default does not deal with alpha values, so faking opacity by fading from the background colour might be good enough). Hopefully most of this can be handled by the menu base class though.

If you agree with above approach I might give it a shot tomorrow.

glipR, 2021-02-23:
(@jtbui20 ^)

jtbui20, 2021-02-23:
Yeah I think I can agree with that approach though I'm not sure whether it would directly impact the performance of the sim?
In the case that there are animations, there will probably be a toggle option for animations in settings.

closed 2021-03-25 by glipR