MelbourneHighSchoolRobotics/ev3sim #280: Windows 10 installation dependency missing

dadalee51, 2021-06-16:
When installing win10 64bit I get this missing dependency error in log:
---------------------------------BEGINNING OF LOG---------------------------------------------------
Collecting ev3sim
  Downloading ev3sim-2.2.0-py3-none-any.whl (2.9 MB)
Collecting opensimplex>=0.3
  Downloading opensimplex-0.3-py3-none-any.whl (15 kB)
Collecting pygame-gui>=0.5.7
  Downloading pygame_gui-0.5.7.tar.gz (1.0 MB)
Collecting debugpy>=1.2.1
  Downloading debugpy-1.3.0-cp39-cp39-win_amd64.whl (4.4 MB)
Collecting sentry-sdk>=0.20.0
  Downloading sentry_sdk-1.1.0-py2.py3-none-any.whl (131 kB)
Collecting mindpile>=0.0.2
  Downloading mindpile-0.0.2-py3-none-any.whl (30 kB)
Collecting python-certifi-win32>=1.6
  Downloading python_certifi_win32-1.6-py2.py3-none-any.whl (7.2 kB)
Collecting luddite>=1.0.1
  Downloading luddite-1.0.1-py2.py3-none-any.whl (9.7 kB)
Collecting python-ev3dev2>=2.1.0.post1
  Downloading python-ev3dev2-2.1.0.post1.tar.gz (77 kB)
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\qq\AppData\Local\EV3Sim\python_embed\python.exe' -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\qq\\AppData\\Local\\Temp\\pip-install-fs45umb2\\python-ev3dev2_27b41d6318b2467d81737413aad4538f\\setup.py'"'"'; __file__='"'"'C:\\Users\\qq\\AppData\\Local\\Temp\\pip-install-fs45umb2\\python-ev3dev2_27b41d6318b2467d81737413aad4538f\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\qq\AppData\Local\Temp\pip-pip-egg-info-moqw735b'
         cwd: C:\Users\qq\AppData\Local\Temp\pip-install-fs45umb2\python-ev3dev2_27b41d6318b2467d81737413aad4538f\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\qq\AppData\Local\Temp\pip-install-fs45umb2\python-ev3dev2_27b41d6318b2467d81737413aad4538f\setup.py", line 2, in <module>
        from git_version import git_version
    ModuleNotFoundError: No module named 'git_version'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/8a/8c/2ddffc70572989acb5d2f60e5e1d7e93f6daf9b05200ac0e80d030e98950/python-ev3dev2-2.1.0.post1.tar.gz#sha256=0cfd69b0e8b6ae2ac8f300d7faea602032d961cc526417554d7da35cd9730949 (from https://pypi.org/simple/python-ev3dev2/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Collecting ev3sim
  Downloading ev3sim-2.1.9-py3-none-any.whl (2.9 MB)
  Downloading ev3sim-2.1.7-py3-none-any.whl (2.8 MB)
Collecting dload>=0.6
  Downloading dload-0.6-py3-none-any.whl (8.1 kB)
Collecting ev3sim
  Downloading ev3sim-2.1.6-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.1.5-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.1.4-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.1.2-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.1.1-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.1.0-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.0.2-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.0.1-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-2.0.0-py3-none-any.whl (2.8 MB)
  Downloading ev3sim-1.6.5-py3-none-any.whl (329 kB)
Collecting pyperclip>=1.8.0
  Using cached pyperclip-1.8.2.tar.gz (20 kB)
Collecting numpy>=1.19.0
  Using cached numpy-1.20.3-cp39-cp39-win_amd64.whl (13.7 MB)
Collecting grpcio-tools>=1.30.0
  Downloading grpcio_tools-1.38.0-cp39-cp39-win_amd64.whl (1.9 MB)
Collecting grpcio>=1.30.0
  Downloading grpcio-1.38.0-cp39-cp39-win_amd64.whl (3.2 MB)
Collecting pygame>=2.0.0
  Downloading pygame-2.0.1-cp39-cp39-win_amd64.whl (5.2 MB)
Collecting ev3sim
  Downloading ev3sim-1.6.4-py3-none-any.whl (329 kB)
  Downloading ev3sim-1.6.3-py3-none-any.whl (329 kB)
  Downloading ev3sim-1.6.2-py3-none-any.whl (328 kB)
  Downloading ev3sim-1.6.1-py3-none-any.whl (328 kB)
  Downloading ev3sim-1.6.0-py3-none-any.whl (328 kB)
  Downloading ev3sim-1.5.3-py3-none-any.whl (328 kB)
  Downloading ev3sim-1.5.2-py3-none-any.whl (327 kB)
  Downloading ev3sim-1.5.1-py3-none-any.whl (327 kB)
  Downloading ev3sim-1.5.0-py3-none-any.whl (327 kB)
  Downloading ev3sim-1.4.0-py3-none-any.whl (324 kB)
  Downloading ev3sim-1.3.3-py3-none-any.whl (318 kB)
  Downloading ev3sim-1.3.2-py3-none-any.whl (318 kB)
  Downloading ev3sim-1.3.1-py3-none-any.whl (317 kB)
  Downloading ev3sim-1.3.0-py3-none-any.whl (317 kB)
  Downloading ev3sim-1.2.3-py3-none-any.whl (313 kB)
  Downloading ev3sim-1.2.2-py3-none-any.whl (313 kB)
  Downloading ev3sim-1.2.1-py3-none-any.whl (311 kB)
  Downloading ev3sim-1.2.0-py3-none-any.whl (310 kB)
  Downloading ev3sim-1.1.0-py3-none-any.whl (66 kB)
  Downloading ev3sim-1.0.9-py3-none-any.whl (64 kB)
  Downloading ev3sim-1.0.8-py3-none-any.whl (64 kB)
ERROR: Cannot install ev3sim==1.0.8, ev3sim==1.0.9, ev3sim==1.1.0, ev3sim==1.2.0, ev3sim==1.2.1, ev3sim==1.2.2, ev3sim==1.2.3, ev3sim==1.3.0, ev3sim==1.3.1, ev3sim==1.3.2, ev3sim==1.3.3, ev3sim==1.4.0, ev3sim==1.5.0, ev3sim==1.5.1, ev3sim==1.5.2, ev3sim==1.5.3, ev3sim==1.6.0, ev3sim==1.6.1, ev3sim==1.6.2, ev3sim==1.6.3, ev3sim==1.6.4, ev3sim==1.6.5, ev3sim==2.0.0, ev3sim==2.0.1, ev3sim==2.0.2, ev3sim==2.1.0, ev3sim==2.1.1, ev3sim==2.1.2, ev3sim==2.1.4, ev3sim==2.1.5, ev3sim==2.1.6, ev3sim==2.1.7, ev3sim==2.1.9 and ev3sim==2.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    ev3sim 2.2.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.9 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.7 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.6 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.5 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.4 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.2 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.1 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.1.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.0.2 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.0.1 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 2.0.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.6.5 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.6.4 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.6.3 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.6.2 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.6.1 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.6.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.5.3 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.5.2 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.5.1 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.5.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.4.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.3.3 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.3.2 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.3.1 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.3.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.2.3 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.2.2 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.2.1 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.2.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.1.0 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.0.9 depends on python-ev3dev2>=2.1.0.post1
    ev3sim 1.0.8 depends on python-ev3dev2>=2.1.0.post1

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies

--------------------------END of LOG-----------------------------

glipR, 2021-06-16:
Thanks for the report @dadalee51 .

It seems based on the error that a file called `git_version.py` is not in the python-ev3dev2 installation setup folder. I'm not sure if this is an ev3sim or python-ev3dev2 bug, or even an issue with pip installing on the embedded python we use.

Would you be able to open `C:\Users\qq\AppData\Local\Temp\pip-install-fs45umb2\python-ev3dev2_27b41d6318b2467d81737413aad4538f` (Or the similar filepath mentioned in the error log if you run the installer again) and check whether `git_version.py` is present?

dadalee51, 2021-06-16:
@glipR thanks for reply, 
the git_version.py was actually in the ev3dev2_xxx directory:
[image]

tried install as admin, made no difference.

glipR, 2021-06-16:
Weird. Do you get a similar error message when running `'C:\Users\qq\AppData\Local\EV3Sim\python_embed\python.exe' -m 'C:\Users\qq\AppData\Local\Temp\pip-install-fs45umb2\python-ev3dev2_27b41d6318b2467d81737413aad4538f\setup.py'`?

(Just opening command prompt and pasting then entering should do the trick)

dadalee51, 2021-06-16:
hmm. couldn't get the hold on to the directory in time, seems like as soon as installation failed all the temp directories got deleted almost instantly. the snapshot you see above was done in a fraction of time.

glipR, 2021-06-16:
Would you be able to try running this attached installer instead? It should have ev3dev2 preinstalled and so hopefully no issues.
Let me know if this is the case so I can just update the release.

Since github doesn't like me uploading .exe files in comments: https://u.pcloud.link/publink/show?code=XZJ2lMXZvdWEFm7DfBFUO9cvcWWH0YnHQgrX

dadalee51, 2021-06-16:
works beautifully now! Thank you!

closed 2021-06-16 by dadalee51

glipR, 2021-06-16:
No worries, thanks for being such a great help. Will update the release .exe files now.